peng-zhihui/L-ink_Card #5: 对nfc有个疑问

Guet-Bantao, 2020-06-21:
任意复制卡片的话，不会冲突吗？
比如多个卡片信息都存在同一扇区

peng-zhihui, 2020-06-21:
仔细看一下硬件工程文件，简单地说目前IC卡的信息是存在每个单独的UID芯片（很便宜大概1~2元一个）上的，ST25有两个作用，一个是能量采集，一个是负责作为手机和MCU通信的桥梁（NFC-PHY）；也就是可以理解为L-ink里面是有多个芯片，共用一个线圈，靠开关切换。之所以这么设计，是因为ST25DV支支持ISO 15693协议，而我们常用的IC卡是ISO 14443协议的，所以并不好直接进行模拟。当然也不排除以后会更新方案完善IC模拟功能，现阶段这个方案对我来说是够用的啦。

peng-zhihui, 2020-06-21:
> 任意复制卡片的话，不会冲突吗？
> 比如多个卡片信息都存在同一扇区

仔细看一下硬件工程文件，简单地说目前IC卡的信息是存在每个单独的UID芯片（很便宜大概1~2元一个）上的，ST25有两个作用，一个是能量采集，一个是负责作为手机和MCU通信的桥梁（NFC-PHY）；也就是可以理解为L-ink里面是有多个芯片，共用一个线圈，靠开关切换。之所以这么设计，是因为ST25DV支支持ISO 15693协议，而我们常用的IC卡是ISO 14443协议的，所以并不好直接进行模拟。当然也不排除以后会更新方案完善IC模拟功能，现阶段这个方案对我来说是够用的啦。

jagobaby33, 2020-06-22:
目前市面上的门禁系统大部分只是读取卡片的uid，所以可以方便复制，如果碰到cpu加密或者hid的iclass卡，应该就没办法了

closed 2020-06-22 by peng-zhihui