peng-zhihui/L-ink_Card #17: 麻烦大佬上传ascii编码的Altium文件

bosssyyyy, 2020-06-22:
麻烦上传一份ascii编码的ad文件，目前我使用的编辑软件不支持ad二进制文件导入。

BH6AOV, 2020-06-22:
您用的立创EDA吧😅我也是，目前在网上找到的官方解决方案没一个成功的，我已经被逼着开始下AD17了233333

bosssyyyy, 2020-06-22:
是(  eda对新手比较友好(((   为了魔改一下两层板去折腾安装ad🌚

BH6AOV, 2020-06-22:
我也是折腾的没办法了(((((((
立创的EDA确实对我这个新手很友好，就是听前辈说立创设计出来的文件有时候去别的软件上会出问题2333333

BH6AOV, 2020-06-22:
http://club.szlcsc.com/article/details_44111_1.html
这个是立创官方给的解决方法，但是我试了之后发现并没有什么卵用🙄

bosssyyyy, 2020-06-22:
那个方案是针对ascii编码下没选择utf-8解决的，这个是二进制编码(好像是)，需要转成ascii。不知道hex直接转有没有用(盲猜没有用，文件头尾什么的也给转了)

BH6AOV, 2020-06-22:
是的，，，我放弃抵抗了，在网上找了个ad17的iso正在下载着呢，3.67GB不挂梯子都下不动（挂了也没快多少）睡醒了应该就差不多下完了

Sorakage033, 2020-06-22:
用AD17转换了一下，转换成ASCII格式会损失一部分信息，直接导入进立创EDA会报DRC错误
链接：https://pan.baidu.com/s/1sIx4CawBqthK2uwkYVPijw 
提取码：dp1h

BH6AOV, 2020-06-22:
> 用AD17转换了一下，转换成ASCII格式会损失一部分信息，直接导入进立创EDA会报DRC错误
> 链接：https://pan.baidu.com/s/1sIx4CawBqthK2uwkYVPijw
> 提取码：dp1h

老哥，稳！

closed 2020-06-22 by bosssyyyy

wangbokun, 2020-06-23:
[image]

立创的零件不全呀，只有三个匹配，14个不行，大佬们，还有其他渠道可以打印板子吗?

BH6AOV, 2020-06-23:
> [image]
> 
> 立创的零件不全呀，只有三个匹配，14个不行，大佬们，还有其他渠道可以打印板子吗?

他说的是有14个型号没匹配上让你手动选一下不是说没有啊23333333

wangbokun, 2020-06-23:
> > [image]
> > 立创的零件不全呀，只有三个匹配，14个不行，大佬们，还有其他渠道可以打印板子吗?
> 
> 他说的是有14个型号没匹配上让你手动选一下不是说没有啊23333333

嗯嗯.  这个倒是的，我看到有些可以选，我是硬件小白，不知道选哪个替换合适，会不会出现不兼容.

BH6AOV, 2020-06-23:
> > > [image]
> > > 立创的零件不全呀，只有三个匹配，14个不行，大佬们，还有其他渠道可以打印板子吗?
> > 
> > 
> > 他说的是有14个型号没匹配上让你手动选一下不是说没有啊23333333
> 
> 嗯嗯. 这个倒是的，我看到有些可以选，我是硬件小白，不知道选哪个替换合适，会不会出现不兼容.

立创里面那些同规格同封装的器件区别基本就是生产厂家和价格的区别23333，只要封装、参数一样理论上是不会出现不兼容的

wangbokun, 2020-06-23:
嗯嗯，好的，那我就准备下个单试试. 电池座那个推荐的产品就有问题，直接给推荐的是5v家用那种电池座，我给换掉了.

[image]

Sorakage033, 2020-06-23:
> [image]
> 
> 立创的零件不全呀，只有三个匹配，14个不行，大佬们，还有其他渠道可以打印板子吗?

打板的话不用pcbdoc，直接在嘉立创上传gerber文件就行了