peng-zhihui/L-ink_Card #22: STM32L051和ST25DV的具体型号

Lauv-who, 2020-06-23:
我在官网看到有很多型号，不知道要买哪个[](url)
[image]
[image]希望能解答下，谢谢

wangbokun, 2020-06-23:
> 我在官网看到有很多型号，不知道要买哪个
> [image]
> [image]希望能解答下，谢谢

大佬，你这个是在哪里下单的呢.我也准备弄一个.

Lauv-who, 2020-06-23:
> > 我在官网看到有很多型号，不知道要买哪个
> 
> > [image]
> 
> > [image]希望能解答下，谢谢
> 
> 
> 
> 大佬，你这个是在哪里下单的呢.我也准备弄一个.

我没下单，不知道能不能买以及购买数量的限制，产地是有中国的

Teng-NUKE, 2020-06-23:
BOM表里有具体型号和封装

Lauv-who, 2020-06-23:
> BOM表里有具体型号和封装

谢谢

peng-zhihui, 2020-06-23:
ST25DV64K，STM32L051K8U6

peng-zhihui, 2020-06-23:
> 我在官网看到有很多型号，不知道要买哪个
> [image]
> [image]希望能解答下，谢谢

ST25DV64K，STM32L051K8U6

Lauv-who, 2020-06-23:
> > 我在官网看到有很多型号，不知道要买哪个
> > [image]
> > [image]希望能解答下，谢谢
> 
> ST25DV64K，STM32L051K8U6

谢谢大佬

doublemarkpro, 2020-06-25:
> > 我在官网看到有很多型号，不知道要买哪个
> > [image]
> > [image]希望能解答下，谢谢
> 
> ST25DV64K，STM32L051K8U6

您好，看原理图RFID芯片用的是ST25DV16K，这里您回复又是64K，以哪个为准呢?

peng-zhihui, 2020-06-25:
> > > 我在官网看到有很多型号，不知道要买哪个
> > > [image]
> > > [image]希望能解答下，谢谢
> > 
> > 
> > ST25DV64K，STM32L051K8U6
> 
> 您好，看原理图RFID芯片用的是ST25DV16K，这里您回复又是64K，以哪个为准呢?

其实都可以，区别只是EEPROM大小不同而已，最好是买64K哈

closed 2020-06-25 by peng-zhihui

doublemarkpro, 2020-06-25:
> > > > 我在官网看到有很多型号，不知道要买哪个
> > > > [image]
> > > > [image]希望能解答下，谢谢
> > > 
> > > 
> > > ST25DV64K，STM32L051K8U6
> > 
> > 
> > 您好，看原理图RFID芯片用的是ST25DV16K，这里您回复又是64K，以哪个为准呢?
> 
> 其实都可以，区别只是EEPROM大小不同而已，最好是买64K哈

好的，元器件已买，pcb自己简单修改了一下已经交去做板，3d打印也交给第三方做去了，另外请教一下，您b站视频中的最后是盖上了一块亚克力外壳吗？这个有无开源尺寸图呢？厚度需要做几个毫米的？谢谢。

Lauv-who, 2020-06-25:
> > > > > 我在官网看到有很多型号，不知道要买哪个
> 
> > > > > [image]
> 
> > > > > [image]希望能解答下，谢谢
> 
> > > > 
> 
> > > > 
> 
> > > > ST25DV64K，STM32L051K8U6
> 
> > > 
> 
> > > 
> 
> > > 您好，看原理图RFID芯片用的是ST25DV16K，这里您回复又是64K，以哪个为准呢?
> 
> > 
> 
> > 其实都可以，区别只是EEPROM大小不同而已，最好是买64K哈
> 
> 
> 
> 好的，元器件已买，pcb自己简单修改了一下已经交去做板，3d打印也交给第三方做去了，另外请教一下，您b站视频中的最后是盖上了一块亚克力外壳吗？这个有无开源尺寸图呢？厚度需要做几个毫米的？谢谢。

大佬在哪里配的单呀，找了很多家都配不齐，谢谢