peng-zhihui/L-ink_Card #24: pcb挖槽位置

Lauv-who, 2020-06-23:
这个在哪开呀[image]

peng-zhihui, 2020-06-23:
IC0和IC1右边那两个框

closed 2020-06-23 by peng-zhihui

Lauv-who, 2020-06-23:
> IC0和IC1右边那两个框

了解，谢谢

Lauv-who, 2020-06-23:
> IC0和IC1右边那两个框

店家说文件没有体现挖空的😥大佬给看下
[image]

wangbokun, 2020-06-24:
我也遇到同样的问题，店家说需要PCB文件上有开槽，才可以，

fengboo, 2020-06-24:
大佬的PCB上只放了一个board cutout的多边形，没用keepout或机械层画边界，工厂只认keepout或机械层边界，多边形没用。

Lauv-who, 2020-06-24:
> 大佬的PCB上只放了一个board cutout的多边形，没用keepout或机械层画边界，工厂只认keepout或机械层边界，多边形没用。

这我也不会画呀😂小白太难了

wangbokun, 2020-06-24:
[image]
不知道这个是不是挖槽成功