peng-zhihui/L-ink_Card #29: 开槽位置请教

wangbokun, 2020-06-24:
[image]
是在这个位置开槽吗？ 看到下面有蓝色的条，可以直接在这里开槽吗？

liangling11411, 2020-06-24:
[image]
是这里（黄笔圈的地方），你直接这样说就行了
[image]
预览图：（圈住的地方开槽）
[image]

Lauv-who, 2020-06-24:
> [image]
> 
> 是这里（黄笔圈的地方），你直接这样说就行了
> 
> [image]
> 
> 预览图：（圈住的地方开槽）
> 
> [image]
> 
> 

在嘉立创下单的话 上传release还是source都可以吗？上传后的具体参数能不能发个图看一下小白不太懂 谢谢

liangling11411, 2020-06-24:
[image]
是release文件夹下的gerber文件夹，把里面的东西打成压缩包（.zip/.rar)
直接发给pcb生产厂家就行了
我的话没有买钢网，如果焊工不好可以整个钢网辅助一下
[image]

Lauv-who, 2020-06-24:
> [image]
> 
> 是release文件夹下的gerber文件夹，把里面的东西打成压缩包（.zip/.rar)
> 
> 直接发给pcb生产厂家就行了
> 
> 我的话没有买钢网，如果焊工不好可以整个钢网辅助一下
> 
> [image]
> 
> 

谢谢您，因为我是小白，所以打算同时下单smt焊接

liangling11411, 2020-06-24:
嗯嗯，玩的开心，没问题了就把issue close一下吧（右下角）

Lauv-who, 2020-06-24:
> 嗯嗯，玩的开心，没问题了就把issue close一下吧（右下角）

谢谢，这个不是我发的issue😀

wangbokun, 2020-06-24:
挖槽那个事情怎么弄的？就给客服说一下就行？不用该文件吗？

liangling11411, 2020-06-24:
> 挖槽那个事情怎么弄的？就给客服说一下就行？不用该文件吗？
先把Gerber文件交给厂家，到时候厂家会回问你开槽的事，你直接说IC0和IC1右边两个框开槽就行了
[image]

wangbokun, 2020-06-24:
好的，明白了，多谢多谢，

closed 2020-06-24 by wangbokun