peng-zhihui/L-ink_Card #31: up主有BOM清单么? :sob:

jkjkiiiii, 2020-06-24:
芯片的型号和拨片还能找到, 电子墨水屏将就找到, 但是很多电阻的阻值、二极管的大小并未提供, 想要复现有点困难. :sob: 求大佬提供BOM清单

closed 2020-06-24 by jkjkiiiii

spideymooon, 2020-06-27:
你好，bom清单有了？

spideymooon, 2020-06-27:
@jkjkiiiii 你好，你有bom清单了吗？能否分析一下

BH6AOV, 2020-06-27:
> @jkjkiiiii 你好，你有bom清单了吗？能否分析一下

BOM清单UP在文件里有发，自己找一下吧

spideymooon, 2020-06-27:
> > @jkjkiiiii你好，你有bom清单了吗？能否分析一下
> 
> BOM清单up在文件里有发，自己找一下吧

好，谢谢

iwtywai, 2020-06-27:
Hardware/release/BOM.xlsx

jkjkiiiii, 2020-06-27:
> 你好，bom清单有了？

虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢:stuck_out_tongue_winking_eye:),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
[image]
如果还嫌图片看的不清可以私聊

spideymooon, 2020-06-27:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊

我的天，太感谢了，就是正愁着有些东西不清楚.蟹蟹！

jikeshen, 2020-06-29:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊

请问拨动开关尺寸是什么，谢谢

jkjkiiiii, 2020-06-29:
> > > 你好，bom清单有了？
> > 
> > 
> > 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> > [image]
> > 如果还嫌图片看的不清可以私聊
> 
> 请问拨动开关尺寸是什么，谢谢

拨动开关我自己去单独买的,视频里截图然后到淘宝问就有