peng-zhihui/L-ink_Card #35: 要不要加个北斗的定位模块，实现自动切换卡片

zmykl, 2020-06-26:
我没细看，想问下mcu有应该还有空闲的io口吧，打算搞个北斗的定位模块，实现靠近相应经纬度坐标之后自动切换到对应的卡片，这样逼格更加高也不需要主动去切换了，最近有空的时候我会做下，到时大家一起交流交流

iwtywai, 2020-06-26:
个人觉得可行，但供电要单独考虑，毕竟eh的电太少了不够用，（就算io不够用的话，都有独立供电了还要什么低能耗）
；）

caicai0, 2020-06-26:
能识别读卡器的不同就好了

iwtywai, 2020-06-26:
前提是读卡器发送的信息不同

XenoAmess, 2020-06-26:
能耗其实 感觉 没几个人在意能耗。。。

zmykl, 2020-06-26:
为了节省功耗肯定要做时间判断的，用加速度传感器或者水银开关来触发，之后唤醒MCU检验时间，每次唤醒后判断下时间是否是大于等于上次切换卡片的十分钟后，是的话获取经纬度进行判断刷新，经纬度差异不大就直接休眠，接近某个坐标点2km内，就切换到对应的卡片上。这样基本上可以兼顾功耗和方便性。获取经纬度刷新的时间有半分钟足够了，实际功耗不会很大。

iwtywai, 2020-06-26:
还可以把纽扣电池换成过充过放保护板接上电池组，再加个usb充电

XenoAmess, 2020-06-26:
> 还可以把纽扣电池换成过充过放保护板接上电池组，再加个usb充电

USB充电是个好东西

BH6AOV, 2020-06-27:
但是如果大门门禁和办公室门禁靠得太近的话定位精度不够也就没法正常工作了啊，2km的范围还是有点太大了，UP视频里两次刷卡的距离明显不超过2km（谁家公司大楼会从大门到办公室建2km的走廊啊）

zmykl, 2020-06-27:
> 但是如果大门门禁和办公室门禁靠得太近的话定位精度不够也就没法正常工作了啊，2km的范围还是有点太大了，UP视频里两次刷卡的距离明显不超过2km（谁家公司大楼会从大门到办公室建2km的走廊啊）

我只是针对我自己的使用需求设计出的切换流程，每个人都有自己的需求，自己实现就行了

BH6AOV, 2020-06-27:
> > 但是如果大门门禁和办公室门禁靠得太近的话定位精度不够也就没法正常工作了啊，2km的范围还是有点太大了，UP视频里两次刷卡的距离明显不超过2km（谁家公司大楼会从大门到办公室建2km的走廊啊）
> 
> 我只是针对我自己的使用需求设计出的切换流程，每个人都有自己的需求，自己实现就行了

如果能实现的话也挺不错的，支持一下

iwtywai, 2020-06-27:
大家一起搞个合作项目？

BH6AOV, 2020-06-27:
我小白，啥都不会（摊手）

iwtywai, 2020-06-27:
我就会画原理图和写简单程序，pcb画的很丑(<_>)

zhangFanML, 2020-06-28:
位置记录容易实现吗?

iwtywai, 2020-06-28:
我的想法是设置三个功能：读卡，记录位置，关联卡和位置信息，其中关联的话应该是要做成一对多的

RishonLi, 2020-06-28:
> 只要一个电池能撑一个月 就完全能接受

钮扣电池不便宜呀

poisonousfish, 2020-06-28:
先不提各个地区精度问题，根据我的使用GPS经验来说，定位并非说一下就能准确的东西，而是需要开机后一段时间通过多次数据进行计算才能变的准确的,所以低功耗就告别了。

15839619751, 2020-06-28:
每天充一次电我都能接受，另外，可以加太阳能啊

ChuangYu1897, 2020-06-29:
> 每天充一次电我都能接受，另外，可以加太阳能啊

太阳能可以有

BH6AOV, 2020-06-29:
233333333越做越复杂了

kmno4-beep, 2020-06-29:
听说开源项目只要有钱有时间就能跟着做，特意注册一个号来看看大家是怎么一起做的（）

Twillight-7, 2020-06-29:
前来观摩大佬，

Twillight-7, 2020-06-29:
前来观摩大佬

iwtywai, 2020-06-29:
先做起来再说，之后再加功能或免用户干预

XenoAmess, 2020-06-29:
还是先保证安全性吧 确定别人拿同样的手机app没法修改你这个固件上的内容吗。。。

pengzi007, 2020-07-01:
1. 不知道压电转换或机械能转换的功率够不够，如果够的话也可以加进去，现在很多门铃的按钮都是可以直接发电的。
2. 另外如果有比较小的指纹模块也可以集成进来，别人捡到的话也没法用。
当然了，加这些新功能的前提是新加的module要足够小。

peng-zhihui, 2020-07-01:
想法没问题，但是实现起来不太可行。北斗和GPS在室内定位都不可靠的，最简单的场景，同一栋楼的两个房间，除非是高精度的室内定位，否则无法区分的。

iwtywai, 2020-07-01:
如果是加上加速度传感器和陀螺仪辅助呢？

peng-zhihui, 2020-07-02:
> 如果是加上加速度传感器和陀螺仪辅助呢？

这属于惯导应用，不要说低功耗设备，就是机器人上的SLAM系统目前都达不到这个效果。惯导应用最成功的是在导弹和火箭上，用的是一颗上百万万的光纤陀螺仪。

zmykl, 2020-07-02:
我觉得这个只要用水银开关或者震动开关触发唤醒就可以了，没必要上加速度传感器的

发自我的小米手机

kururu111, 2020-07-02:
能不能加个刷一次卡就自己切换成下一张的功能，因为每个人每天的行程差不多都是固定的，只要按顺序让卡片自己切换卡的类型，在app上可以编辑换卡的顺序应该可以吧?

iwtywai, 2020-07-02:
这个可以有，改下程序的事

or2me, 2020-07-06:
> 能不能加个刷一次卡就自己切换成下一张的功能，因为每个人每天的行程差不多都是固定的，只要按顺序让卡片自己切换卡的类型，在app上可以编辑换卡的顺序应该可以吧?或者在linkcard加上时钟模块应该可以玩出更多花样。

就你这个比较靠谱 不要光想着空间维度 时间维度明显简单多了啊 毕竟大部分人的工作生活是规律的

imyip, 2020-07-08:
我觉得滑轮切换, 显示器显示当前卡的编号就好了 ...

hsuehnan, 2020-07-16:
> 我觉得滑轮切换, 显示器显示当前卡的编号就好了 ...

我觉得也是，应该不用多个按键吧，而且功耗这个，不会太介意的，就是不知道交通卡能不能用

iwtywai, 2020-07-16:
我觉得我太难了，到现在连个门禁卡都没有破开，用的pn532

doskoi, 2020-07-18:
GPS相关的芯片基本上只有晴天的户外才有信号

QuietCao, 2020-07-21:
> 我觉得滑轮切换, 显示器显示当前卡的编号就好了 ...

我觉得连显示器都不需要，弄几个灯来显示编号就行了

Glover120213, 2020-07-21:
敬爱的稚晖君：
   我是一名来自常州的初中生，非常喜欢您的项目；
   对于这个问题，我觉得可以增加北斗模块，但是如果运用在自动换卡片上好像有点大材小用，可以针对北斗定位开发项目；
   其次，我有一个想法，就是加入时钟模块，然后将卡片给老人和小孩（这样他们可以利用NFC自己乘车出去玩，还能用卡片进小区门，还可以用它看时间，走丢了有北斗定位）；
   最后，有一些朋友提出可以去掉电子屏，我觉得如果想在这个卡片上开发一些新的有意思的功能，显示器还是要的；如果去掉显示屏，那是不是可以考虑用一开始不加电池的方案，做成精简版？
   祝您身体健康，万事如意！
   此致，
敬礼！
By Gs―Your friend all the time!
Tips:发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

Glover120213, 2020-07-21:
敬爱的稚晖君：
   我是一名来自常州的初中生，非常喜欢您的项目；
   对于这个问题，我觉得可以增加北斗模块，但是如果运用在自动换卡片上好像有点大材小用，可以针对北斗定位开发项目；
   其次，我有一个想法，就是加入时钟模块，然后将卡片给老人和小孩（这样他们可以利用NFC自己乘车出去玩，还能用卡片进小区门，还可以用它看时间，走丢了有北斗定位）；
   最后，有一些朋友提出可以去掉电子屏，我觉得如果想在这个卡片上开发一些新的有意思的功能，显示器还是要的；如果去掉显示屏，那是不是可以考虑用一开始不加电池的方案，做成精简版？
   祝您身体健康，万事如意！
   此致，
敬礼！

zhangFanML, 2020-07-21:
有趣

发送自 Windows 10 版邮件应用

发件人: Glover120213
发送时间: 2020年7月21日 13:07
收件人: peng-zhihui/L-ink_Card
抄送: zhangFanML; Comment
主题: Re: [peng-zhihui/L-ink_Card] 要不要加个北斗的定位模块，实现自动切换卡片 (#35)

敬爱的稚晖君：
我是一名来自常州的初中生，非常喜欢您的项目；
对于这个问题，我觉得可以增加北斗模块，但是如果运用在自动换卡片上好像有点大材小用，可以针对北斗定位开发项目；
其次，我有一个想法，就是加入时钟模块，然后将卡片给老人和小孩（这样他们可以利用NFC自己乘车出去玩，还能用卡片进小区门，还可以用它看时间，走丢了有北斗定位）；
最后，有一些朋友提出可以去掉电子屏，我觉得如果想在这个卡片上开发一些新的有意思的功能，显示器还是要的；如果去掉显示屏，那是不是可以考虑用一开始不加电池的方案，做成精简版？
祝您身体健康，万事如意！
此致，
敬礼！

printlin, 2020-07-24:
我可以做软件开发相关的工作，STM32和APP包括服务器程序都可以，要是有硬件大佬正在继续尝试项目的延续，可以联系我。

iwtywai, 2020-07-24:
> 我可以做软件开发相关的工作，STM32和APP包括服务器程序都可以，要是有硬件大佬正在继续尝试项目的延续，可以联系我。

我有一个问题想要请教，可以加一下好友吗？我的qq是2745076015

jiangyou233, 2020-07-24:
> 敬爱的稚晖君：
> 我是一名来自常州的初中生，非常喜欢您的项目；
> 对于这个问题，我觉得可以增加北斗模块，但是如果运用在自动换卡片上好像有点大材小用，可以针对北斗定位开发项目；
> 其次，我有一个想法，就是加入时钟模块，然后将卡片给老人和小孩（这样他们可以利用NFC自己乘车出去玩，还能用卡片进小区门，还可以用它看时间，走丢了有北斗定位）；
> 最后，有一些朋友提出可以去掉电子屏，我觉得如果想在这个卡片上开发一些新的有意思的功能，显示器还是要的；如果去掉显示屏，那是不是可以考虑用一开始不加电池的方案，做成精简版？
> 祝您身体健康，万事如意！
> 此致，
> 敬礼！

首先 不加电池 哪里来的供电那  不要来太阳能啥的啦  时钟有电池供电的话 可以一直跑的 公交卡很多都是CPU卡 不能复制

iwtywai, 2020-07-24:
把公交卡里的芯片拆除来焊上去🤤

Glover120213, 2020-07-24:
我是看稚晖在视频里的第一个方案，直接利用芯片供电采集，当时不是因为供电不足不成立么，现在去掉了大部分用电器，是不是可以考虑重新使用那个方案来达到简化结构
By Gs―Your friend all the time!
Tips:发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

iwtywai, 2020-07-24:
现在整个供电不仍然是energy harvesting吗？电池是给墨水屏用的

jiangyou233, 2020-07-24:
> 现在整个供电不仍然是energy harvesting吗？电池是给墨水屏用的

不是 看原理图 电池通过TPS61221 转为3.3V

jiangyou233, 2020-07-24:
开始通过线圈充电给供电的  后来发现读卡器供电大小不一，充电时间也有长短 所以放弃  我没记错的话

iwtywai, 2020-07-24:
那不就是energy harvesting吗...

Fir-Geek, 2020-07-24:
> 那不就是energy harvesting吗...

看下原理图 真相会浮出水面的

Glover120213, 2020-07-24:
好像是哎，但是我觉得可以改得大众化一点……那电池最好改造成充电电池或者设计得便于拆卸那种吧……不然这么好的东西，做好送给一些不懂电路的人，回头长久使用的话对于他们来说好像有点麻烦
By Gs―Your friend all the time!
Tips:发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

iwtywai, 2020-07-24:
可以考虑一下过充过放保护板+可充电电池+usb，只要留个口就可以

NBchitu, 2020-08-15:
我提供下自己的思路：
1、根据时间段来自动学习并存储刷卡习惯，前面学习一段时间后设备就记录下你的使用习惯，加入点简单的AI学习的功能，如根据时间、当天第几次刷卡等特征值来对当前刷卡机做一个预测；
2、针对1中预测失败的情况，设备马上切换到预测概率第2大的卡进行刷卡操作，也就是实现自动轮询枚举
以上是否可行大家可以探讨下。

iwtywai, 2020-08-15:
想法是好的，但是用stm训练ai不现实，根据st官方的网站https://www.st.com/content/st_com/zh/about/innovation---technology/artificial-intelligence.html，arm cortex m是只能运行预训练的神经网络的

iwtywai, 2020-08-15:
如果只是根据时间来计算的话倒是有可能实现

huarongze, 2020-10-11:
> 
> 
> 我可以做软件开发相关的工作，STM32和APP包括服务器程序都可以，要是有硬件大佬正在继续尝试项目的延续，可以联系我。

多年的硬件工程师，一起讨论下项目的延续，Q422968316

SNGDude, 2020-12-31:
有点远了吧兄弟们，pcb图好说，关键这么叠罗汉不是成手机了。就这么大接触面积一块eink还要上太阳能，还要加北斗定位，室内京东那你用wifi定位+Celluar，这耗电量。