peng-zhihui/L-ink_Card #38: 关于导出Gerber

BH6AOV, 2020-06-27:
在立创上用UTF-8存的ASCII的pcb设计文件准备导出gerber去打样了，但是发现网络和DRC有一大堆报错，是直接忽略就行吗
[image]
[image]

iwtywai, 2020-06-27:
在Hardware文件夹有up给的gerber啊

BH6AOV, 2020-06-27:
> 在Hardware文件夹有up给的gerber啊

立创打不开的

BH6AOV, 2020-06-27:
> 在Hardware文件夹有up给的gerber啊

具体见#17

iwtywai, 2020-06-27:
找个打板的把gerber扔过去就行

BH6AOV, 2020-06-27:
> 找个打板的把gerber扔过去就行

我试试直接扔下单助手上

iwtywai, 2020-06-27:
记得打包.zip

BH6AOV, 2020-06-27:
> 记得打包.zip

妥

BH6AOV, 2020-06-27:
> 找个打板的把gerber扔过去就行

立创那边直接gerber解析超时了

iwtywai, 2020-06-27:
可以多试几次，或者直接在淘宝找一家打板，我是在淘宝打的

BH6AOV, 2020-06-27:
> 可以多试几次，或者直接在淘宝找一家打板，我是在淘宝打的

多少钱啊。。。我能给PCB的预算非常少

iwtywai, 2020-06-27:
我打的无铅黑油，50元5张板

minemc24236, 2020-06-27:
立创5元的，50打针土豪

minemc24236, 2020-06-27:
我是像楼主你这么打的，我看报错是地线和天线，也就没管而且就5元吗，大不了再打一版awa

iwtywai, 2020-06-27:
（我当时打板子gerber文件up还没传，所以用不了jlc）

ZKLgayhub, 2020-06-27:
1.图1,底层贴片层多出来的开窗要在底层阻焊层开上窗吗 2，图2，箭头位的锣槽输出文件仅在底层字符层有框，就按底层字符层框的大小5*4MM锣出方槽吧

求问，是PCB厂家发过来的，本人小白完全不懂

[image]
[image]

BH6AOV, 2020-06-27:
> 1.图1,底层贴片层多出来的开窗要在底层阻焊层开上窗吗 2，图2，箭头位的锣槽输出文件仅在底层字符层有框，就按底层字符层框的大小5*4MM锣出方槽吧
> 
> 求问，是PCB厂家发过来的，本人小白完全不懂
> 
> [image]
> [image]

@peng-zhihui

BH6AOV, 2020-06-27:
> 1.图1,底层贴片层多出来的开窗要在底层阻焊层开上窗吗 2，图2，箭头位的锣槽输出文件仅在底层字符层有框，就按底层字符层框的大小5*4MM锣出方槽吧
> 
> 求问，是PCB厂家发过来的，本人小白完全不懂
> 
> [image]
> [image]

#29

peng-zhihui, 2020-06-28:
解析超市不用管，是嘉立创软件的问题，直接提交确认就行

closed 2020-06-28 by peng-zhihui