peng-zhihui/L-ink_Card #44: 那磁珠那个是用磁珠呢还是用电感呢> 你好，bom清单有了？

782884341, 2020-06-29:
> 你好，bom清单有了？

虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢:stuck_out_tongue_winking_eye:),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
[image]
如果还嫌图片看的不清可以私聊

_Originally posted by @jkjkiiiii in https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239_

wangbokun, 2020-06-29:
你是在哪个站买的元件.

jiangyou233, 2020-06-29:
> 你是在哪个站买的元件.

哈哈 我也想知道 感觉那个NFC芯片有点小贵

jkjkiiiii, 2020-06-29:
磁珠

doublemarkpro, 2020-06-30:
这个ST25DV64K是真难买，淘宝几家都是7-10天，索性就在立创商场买了

yelvlab, 2020-07-01:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊
> 
> _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

FPC座子是抽屉式还是翻盖式的

Lauv-who, 2020-07-01:
> > 你好，bom清单有了？
> 
> 
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢:stuck_out_tongue_winking_eye:),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> 
> [image]
> 
> 如果还嫌图片看的不清可以私聊
> 
> 
> 
> _Originally posted by @jkjkiiiii in https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239_

这个备注貌似是我做的 哈哈

jkjkiiiii, 2020-07-01:
> > > 你好，bom清单有了？
> > 
> > 
> > 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> > [image]
> > 如果还嫌图片看的不清可以私聊
> > _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_
> 
> 这个备注貌似是我做的 哈哈

Thanks♪(･ω･)ﾉ感谢, 借用一下你的备注. 看来我们在淘宝选择了同一家店铺, 他们家电阻电容都是100个起卖, 我跟他们说能不能我50个电阻你50个电阻, 店员说不可以.hhhhhh

Lauv-who, 2020-07-01:
> Thanks♪(･ω･)ﾉ感谢, 借用一下你的备注. 看来我们在淘宝选择了同一家店铺, 他们家电阻电容都是100个起卖, 我跟他们说能不能我50个电阻你50个电阻, 店员说不可以.hhhhhh

是的一套下来好几百 我已经焊坏三个板子了 😥

im12, 2020-07-01:
> 是的一套下来好几百 我已经焊坏三个板子了 😥

算下来一套成本93吧，我已经做完了

Lauv-who, 2020-07-01:
> 算下来一套成本93吧，我已经做完了

大佬怎么焊的

yelvlab, 2020-07-01:
> 算下来一套成本93吧，我已经做完了

大佬，你的24p fpc是抽屉式的还是下翻盖的，然后ST25DV是立创商城还是淘宝买的啊，然后UID芯片呢

im12, 2020-07-01:
> 大佬怎么焊的

锡膏加风枪，和稚晖大佬视频里的差不多吧，我没做钢网，直接用针筒点的锡膏

Lauv-who, 2020-07-01:
> 锡膏加风枪，和稚晖大佬视频里的差不多吧，我没做钢网，直接用针筒点的锡膏

我买了焊台 6脚的拨片开关焊的连锡 坏了3板子 我去买个风枪 谢谢大佬

im12, 2020-07-01:
> 大佬，你的24p fpc是抽屉式的还是下翻盖的，然后ST25DV是立创商城还是淘宝买的啊，然后UID芯片呢

24p我用的是抽屉式的，理论上翻盖的也可以用，只要是下接的都行；ST25DV是淘宝上一家家问过来终于有一家有现货就买了，04K到64K都可以用，up的程序里只用了其中的mailbox，没用eprom，所以大小无所谓，up好像是在贸泽电子买的，我上去看了确实有货还不贵，可以考虑；最后UID芯片的话，我没有写卡设备也不想做啥卡模拟，就直接买了几颗NTAG213的芯片做nfc标签了，实现功能的话比如一刷直接连上我家WiFi这种

peng-zhihui, 2020-07-01:
多谢分享哈

laucong, 2020-07-23:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊
> 
> _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

你好，想问下波动开关的型号是什么呢，淘宝买的时候店家说需要提供型号~

GaoChen-git, 2020-08-07:
> 多谢分享哈
UP，你的UID芯片型号是啥呀？

GaoChen-git, 2020-08-07:
> > 多谢分享哈
> > UP，你的UID芯片型号是啥呀？

然后写卡还需要买其他模块吗？有啥步骤方法？

mdhuang555, 2020-09-22:
> > > > 你好，bom清单有了？
> > > 
> > > 
> > > 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢stuck_out_tongue_winking_eye),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> > > [image]
> > > 如果还嫌图片看的不清可以私聊
> > > _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_
> > 
> > 
> > 这个备注貌似是我做的 哈哈
> 
> Thanks♪(･ω･)ﾉ感谢, 借用一下你的备注. 看来我们在淘宝选择了同一家店铺, 他们家电阻电容都是100个起卖, 我跟他们说能不能我50个电阻你50个电阻, 店员说不可以.hhhhhh

可以买个电容电阻样品册，这样就什么型号的电阻电容都有了

ice-shimmer, 2020-10-20:
> 算下来一套成本93吧，我已经做完了

大佬我没啥焊接基础，某宝有没有这种代购材料+焊接的服务啊

iwtywai, 2020-10-20:
可以在嘉立创下smt订单的

ice-shimmer, 2020-10-20:
> 可以在嘉立创下smt订单的

搜嘎，多谢大佬

jkjkiiiii, 2020-10-20:
不会很便宜就是了:laughing:

PANZERUI, 2020-11-26:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊
> 
> _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

怎么联系您呢，有没有VX啥的

SaleJuice, 2021-04-07:
我想知道水墨屏是什么型号的呢？有人知道吗？

ruinanzhu, 2021-04-07:
微雪1.54寸两色墨水屏-------- 原始邮件 --------发件人： SaleJuice ***@***.***>日期： 2021年4月7日周三 10:56收件人： peng-zhihui/L-ink_Card ***@***.***>抄送： Subscribed ***@***.***>主    题： Re: [peng-zhihui/L-ink_Card] 那磁珠那个是用磁珠呢还是用电感呢> 你好，bom清单有了？ (#44)
我想知道水墨屏是什么型号的呢？有人知道吗？

liu-x-sh, 2021-04-15:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考
> [image]
> 如果还嫌图片看的不清可以私聊
> 
> _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

想问下拨盘电位器有推进链接或者型号嘛？为什么我淘宝买了几个都不合适？

Ceoha, 2021-05-09:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单，但是很多还是写的明显，对于我们这种没自己采购过原件的人很不友好（谁让我想做呢），我这里有一份比较详细的清单清单+补充说明，可以 送给你作为参考如果还嫌图片看的不清可以私聊
> [image]
> 
> _最初由@jkjkiiiii[发表](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)于[＃31（评论）](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

有人做成功了吗

OCEANOUXIN, 2021-11-25:
> > 你好，bom清单有了？
> 
> 虽然有up主发了清单,但是很多还是写的不清楚, 对于我们这种没自己采购过原件的人很不友好(谁让我想做呢😜),我这里有一份比较详细的清单列表+补充说明,可以送给你作为参考 [image] 如果还嫌图片看的不清可以私聊
> 
> _Originally posted by @jkjkiiiii in [#31 (comment)](https://github.com/peng-zhihui/L-ink_Card/issues/31#issuecomment-650559239)_

请问有文档吗，感谢