peng-zhihui/L-ink_Card #45: BOM清单购买问题

wangbokun, 2020-06-29:
[image]
还有九款元件没找到在哪里买，哪位大神给知道.

wangbokun, 2020-06-29:
电池扣我的链接好像是错的，上面找的是个圆形的， 看别人发的是这款:
[image]

购买链接如下:

https://item.taobao.com/item.htm?spm=a1z10.3-c.w4002-21822726391.28.7644315aWrBTWI&id=592903534595

wangbokun, 2020-06-29:
[image]

Blueanr, 2020-06-29:
1、墨水屏 200x200单色微雪4、
10、 立创商城难道没有0Ω0402的电阻卖吗

jiangyou233, 2020-06-29:
r9怎么破  不接吗？

Blueanr, 2020-06-29:
1、墨水屏 200x200单色微雪
3、BOM单里0603的电容我只看到有一个C12 4.7uf
4、STM32L0随便买
8、立创商城（淘宝）难道没有ST25DV16K-JFR6D3卖吗
9、开关
10、 立创商城（淘宝）难道没有0Ω0402的电阻卖吗
11、4pin插座
12、立创商城（淘宝）难道没有68uH2520的电感卖吗
14、立创商城（淘宝）难道没有TPS61221DCKT卖吗

yelvlab, 2020-06-30:
> 1，墨水屏200x200单色微雪
> 3，BOM单里0603的电容我只看到有一个C12 4.7uf
> 4，STM32L0随便买
> 8，立创商城（淘宝）难道没有ST25DV16K-JFR6D3卖吗
> 9，开关
> 10 ，立创商城（淘宝）难道没有0Ω0402的电阻卖吗
> 11、4pin插座
> 12，立创商城（淘宝）难道没有68uH2520的电感卖吗
> 14，立创商城（淘宝）难道没有TPS61221DCKT卖吗？

分享一下我的精力
> 8 发货都很慢，倒不是找不到
> 9 我是tb搜贴片拨轮，然后找的6脚的
> 12 高频电感我是tb买的

ruinanzhu, 2020-07-01:
有没有大佬做完出售一个

jiangyou233, 2020-07-01:
物料成本将近70   还没算人工

ruinanzhu, 2020-07-01:
> 物料成本将近70 还没算人工

我自己做5个得快500成本，还不包括找同事帮忙焊板子请吃饭的钱

Sloan-Lin, 2020-07-01:
话说找嘉立创做PCB和SMT的话整个板做好要多少钱？（纯小白发言）

csr865, 2020-07-02:
bom表中有几个参数不完整怎么回事

csr865, 2020-07-02:
有5个没有具体封装型号

BH6AOV, 2020-07-02:
这是我在立创的购买清单，有的型号不太一样但是是理论可以通换的，成本已经尽我所能压到最低了233333

购买类型 | 商品编号 | 物料编码 | 商品分类 | 名称 | 商品型号 | 品牌 | 封装规格 | 单个毛重 | 购买数量 | 商品单价 | 金额
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
常规 | C557591 |   | 轻触开关 | 轻触开关 3*4*2 两脚小龟不带柱 250gf 黑色塑胶按钮 | TS342A2P-WZ | SHOU HAN(首韩) | SMD | 0.000126 | 20 | ￥0.184220 | ￥3.68
常规 | C262567 |   | FFC，FPC连接器 | FPC 0.5MM 24P 拉抽式 H2.0mm 下接 | AFC07-S24FCA-00 | JS(钜硕电子) | P=0.5mm | 0.000570 | 10 | ￥0.568392 | ￥5.68
常规 | C541858 |   | 排针排母 | 排针 间距2.0mm 4P 单排直插 | PZ200V-11-04P | XFCN(台湾兴飞) | Through Hole,P=2.00mm | 0.000127 | 50 | ￥0.088426 | ￥4.42
常规 | C487769 |   | MOS(场效应管) | UTC   场效应管 | 2N7002KG-AE2-R | UTC(友顺) | SOT23-3 | 0.000017 | 20 | ￥0.132145 | ￥2.64
常规 | C64885 |   | 肖特基二极管 | B5819W   SL | B5819W | MDD | SOD-123 | 0.000035 | 40 | ￥0.106373 | ￥4.25
常规 | C401112 |   | 发光二极管 | LED贴片0402高亮蓝色光 | MHT136UBCT | MEIHUA(美华科技) | 0402 | 0.000020 | 10 | ￥0.227351 | ￥2.27
常规 | C70378 |   | 电池盒/电池座 | CR2032电池 镀金贴片卧式底座 | BS-8-1 | Q&J | 贴片 | 0.001300 | 10 | ￥1.0728 | ￥10.73
常规 | C5407 |   | 电池 | 3V  纽扣电池 | CR2032 | 天球 | 20×3.2 | 0.004280 | 10 | ￥0.745056 | ￥7.45
常规 | C140214 |   | 贴片电阻 | 10KΩ(103)   ±5% | RC-02W103JT | FH(风华) | 0402 | 0.000022 | 100 | ￥0.003227 | ￥0.32
常规 | C140224 |   | 贴片电阻 | 0Ω(000)   ±5% | RC-02000JT | FH(风华) | 0402 | 0.000022 | 100 | ￥0.003476 | ￥0.35
常规 | C83062 |   | 贴片电容 | 10nF(103) ±10% 50V | 0402B103K500 | 华新科(Walsin) | 0402 | 0.000010 | 100 | ￥0.008623 | ￥0.86
常规 | C99226 |   | 贴片电容 | 4.7uF(475) ±10% 16V | 0603B475K160NT | FH(风华) | 0603 | 0.000030 | 20 | ￥0.212655 | ￥4.25
常规 | C387941 |   | 贴片电容 | 100nF(104) ±10% 10V | 0402B104K100CT | 华新科(Walsin) | 0402 | 0.000010 | 100 | ￥0.009396 | ￥0.94
常规 | C106833 |   | 贴片电容 | 10uF(106) ±20% 6.3V | 0402X106M6R3 | 华新科(Walsin) | 0402 | 0.000010 | 20 | ￥0.197150 | ￥3.94
常规 | C144761 |   | 贴片电阻 | 3KΩ(3001)   ±1% | AC0402FR-073KL | YAGEO(国巨) | 0402 | 0.000009 | 100 | ￥0.008709 | ￥0.87
常规 | C379996 |   | 贴片电感 | 4.7uH   ±20% | MIP20124R7MBP | INPAQ(台湾佳邦) | 0805 | 0.000044 | 20 | ￥0.213947 | ￥4.28
常规 | C25545 |   | 贴片电阻 | 20KΩ   ±5% | 0402WGJ0203TCE | UNI-ROYAL(厚声) | 0402 | 0.000009 | 100 | ￥0.003126 | ￥0.31
常规 | C25930 |   | 贴片电阻 | 1.5KΩ   ±5% | 0402WGJ0152TCE | UNI-ROYAL(厚声) | 0402 | 0.000009 | 100 | ￥0.003279 | ￥0.33
常规 | C107372 |   | 贴片电容 | 1uF(105) ±10% 6.3V | CL05A105KQ5NNNC | SAMSUNG(三星) | 0402 | 0.000013 | 100 | ￥0.013405 | ￥1.34
常规 | C279987 |   | 贴片电阻 | 1KΩ(102)   ±5% | CR-02JL6----1K | 光颉(Viking) | 0402 | 0.000009 | 100 | ￥0.001970 | ￥0.20
常规 | C166647 |   | 贴片电阻 | 3Ω(3R0)   ±5% | RTT023R0JTH | RALEC(旺诠) | 0402 | 0.000009 | 100 | ￥0.003609 | ￥0.36
赠品 | C500622 |   | 其余赠品 | SOP16封装音频功率放大器方案验证板 | SOP16封装音频功率放大器方案验证板 | LC(立创) | - | 0.003830 | 1 | ￥0 | ￥0
  |   |   |   |   |   |   |   |   |   | 合计 | ￥59.47

pcwuyu, 2020-07-06:
有没有大佬做好成品出售的？有的话联系我呗，一个就成

closed 2020-07-13 by wangbokun

halflesson, 2020-07-23:
拨动的开关型号是啥样的呀，淘宝怎么搜索呢

iwtywai, 2020-07-23:
在risym家搜贴片拨轮就行

xiaxichen, 2023-09-23:
你好 C83062 C99226 的替代品是什么呢？

Rulsion, 2023-09-23:
已收到，放心

Blueanr, 2023-09-23:
> 你好 C83062 C99226 的替代品是什么呢？

随便找两个封装一样参数相同的电容就行了

xiaxichen, 2023-10-28:
请问一下这个芯片买哪个呢？

> > 你好 C83062 C99226 的替代品是什么呢？
> 
> 随便找两个封装一样参数相同的电容就行了

Blueanr, 2023-10-28:
[image]
根据这两个参数随便买

Blueanr, 2023-10-28:
> 请问一下这个芯片买哪个呢？
> 
> > > 你好 C83062 C99226 的替代品是什么呢？
> > 
> > 
> > 随便找两个封装一样参数相同的电容就行了

[image]
根据这两个参数随便买
0402封装10nf电容和0603封装4.7uf电容

xiaxichen, 2023-10-28:
> > 请问一下这个芯片买哪个呢？
> > > > 你好 C83062 C99226 的替代品是什么呢？
> > > 
> > > 
> > > 随便找两个封装一样参数相同的电容就行了
> 
> [image] 根据这两个参数随便买 0402封装10nf电容和0603封装4.7uf电容

对这个我买到了 谢谢 好心人， 主要是除了这些电容我还需要买什么呢？