peng-zhihui/L-ink_Card #49: 求助帖

jimianjun, 2020-06-30:
谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重阻碍，可以联系下我不，wx就是用户名，求加

lewin0301, 2020-06-30:
电子元件淘宝买啊

jimianjun, 2020-06-30:
> 电子元件淘宝买啊

我知道呀，太难了，要么没有，要么不确定，可以教我走下路不，你搞成了没，教下我吧

lewin0301, 2020-06-30:
> > 电子元件淘宝买啊
> 
> 
> 
> 我知道呀，太难了，要么没有，要么不确定，可以教我走下路不，你搞成了没，教下我吧

我也是小白，我是按照bom表挨个搜的

jimianjun, 2020-06-30:
> > > 电子元件淘宝买啊
> > 
> > 
> > 我知道呀，太难了，要么没有，要么不确定，可以教我走下路不，你搞成了没，教下我吧
> 
> 我也是小白，我是按照bom表挨个搜的

然后呢，都到了吗？商家没疑问什么吗？你是搞回来自己贴片？

lewin0301, 2020-06-30:
> > > > 电子元件淘宝买啊
> 
> > > 
> 
> > > 
> 
> > > 我知道呀，太难了，要么没有，要么不确定，可以教我走下路不，你搞成了没，教下我吧
> 
> > 
> 
> > 我也是小白，我是按照bom表挨个搜的
> 
> 
> 
> 然后呢，都到了吗？商家没疑问什么吗？你是搞回来自己贴片？

我加你微信了，我是才买的电子元件，今天刚打的板

liangling11411, 2020-06-30:
> 谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重矛盾，可以联系下我不，wx就是用户名，求加
可以去tb搜一下配单的商家，你直接把bom表发给客服就行了，如果有元件不知道型号可以看看以前的issue，如果客服说没有哪个元件（比如那个st25dv）就自己买，很简单的

jimianjun, 2020-06-30:
> > > > > 电子元件淘宝买啊
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > 我知道呀，太难了，要么没有，要么不确定，可以教我走下路不，你搞成了没，教下我吧
> > 
> > 
> > > 
> > 
> > 
> > > 我也是小白，我是按照bom表挨个搜的
> > 
> > 
> > 然后呢，都到了吗？商家没疑问什么吗？你是搞回来自己贴片？
> 
> 我加你微信了，我是才买的电子元件，今天刚打的板

那你这还新手啊

jimianjun, 2020-06-30:
> > 谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重矛盾，可以联系下我不，wx就是用户名，求加
> > 可以去tb搜一下配单的商家，你直接把bom表发给客服就行了，如果有元件不知道型号可以看看以前的问题，如果客服说没有这个元素（某些那个st25dv）就自己买，很简单的

元件应该没问题能凑全，回来自己贴片不会哈，还有后续的写固件，还有用什么数据线连板子，想学习无从下手，😭

liangling11411, 2020-06-30:
> > > 谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重矛盾，可以联系下我不，wx就是用户名，求加
> > > 可以去tb搜一下配单的商家，你直接把bom表发给客服就行了，如果有元件不知道型号可以看看以前的问题，如果客服说没有这个元素（某些那个st25dv）就自己买，很简单的
> 
> 元件应该没问题能凑全，回来自己贴片不会哈，还有后续的写固件，还有用什么数据线连板子，想学习无从下手，

可以去嘉立创开个钢网，也不复杂（就是有点小贵）写固件可以看看https://github.com/peng-zhihui/L-ink_Card/issues/30 和 https://github.com/peng-zhihui/L-ink_Card/issues/6

Blueanr, 2020-06-30:
？？？纯小白买回来了会焊？？0402都没米粒大你有风枪吗？？

liangling11411, 2020-06-30:
> ？？？纯小白买回来了会焊？？0402都没米粒大你有风枪吗？？

啊这，也是（虽然我有）

lewin0301, 2020-06-30:
> ？？？纯小白买回来了会焊？？0402都没米粒大你有风枪吗？？

我公司有啊

TaXue-DuBa, 2020-07-03:
> 谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重阻碍，可以联系下我不，wx就是用户名，求加

板子制作去嘉立创，每个月两次，每次5元5块，元器件可以立创商城或者淘宝

BH6AOV, 2020-07-03:
<a href="https://github.com/peng-zhihui/L-ink_Card/issues/45#issuecomment-653030940">我在立创商城的购买清单都在这了</a>，另外提一句，建议CR2032去1688买，比商城里便宜

lvshuy, 2020-07-09:
> [我在立创商城的购买清单都在这了](https://github.com/peng-zhihui/L-ink_Card/issues/45#issuecomment-653030940)，另外提一句，建议CR2032去1688买，比商城里便宜

感谢感谢，很详细了~

467815891a, 2020-07-20:
某宝有卖成品的呀，你这也太小白了

whthend, 2020-08-01:
为啥要手动焊接呢？嘉立创不是有smt贴片的选项嘛？

TaXue-DuBa, 2020-08-06:
> 为啥要手动焊接呢？嘉立创不是有smt贴片的选项嘛？

。。贵啊，而且未必所有原件都有SMT啊，有些东西还是要自己贴的

dorlolo, 2020-10-13:
我外壳在淘宝订的带挂孔版本，买了两套。店铺老板给了我半价50元（无喷漆纯白色）。感觉老板人不错，帮老板留个微信q2576695321，加她好友也给半价优惠（声明：最后请从淘宝下单，毕竟网上认识的，不要直接汇款）。

另外问一下 2520-L 68uH和6引脚拨动开关哪家店铺都有货的？我不想多花邮费，最好一家店铺购买。预算一直在蹭蹭蹭网上涨。。

huajianmao, 2020-10-19:
https://item.taobao.com/item.htm?spm=a230r.1.14.1.40c422f1amQmIO&id=623134077131&ns=1&abbucket=11#detail

JH0220, 2022-04-30:
> > > > 谁那有购买过的商铺的吗？我是个纯小白，元件都没接触过，谁可以指导下我吗，我无从下手啊，目前外壳和屏幕购买了，板子制作，元件购买，遇到了重重矛盾，可以联系下我不，wx就是用户名，求加
> > > > 可以去tb搜一下配单的商家，你直接把bom表发给客服就行了，如果有元件不知道型号可以看看以前的问题，如果客服说没有这个元素（某些那个st25dv）就自己买，很简单的
> > 
> > 
> > 元件应该没问题能凑全，回来自己贴片不会哈，还有后续的写固件，还有用什么数据线连板子，想学习无从下手，
> 
> 可以去嘉立创开个钢网，也不复杂（就是有点小贵）写固件可以看看#30 和 #6

您好 想请教一下 嘉立创打板这个尺寸是多少啊