peng-zhihui/L-ink_Card #56: l-link_card，ic卡问题

jimianjun, 2020-07-04:
pcb板子上挖的那两个洞是做什么用的，st25d芯片可以直接模拟ic卡吗？还是需要在那两个洞里面添加两个ic芯片吗？可以添加几张卡进里面？有人懂吗？问了好几个人都不知道

iwtywai, 2020-07-04:
目前up的版本是添加两张卡，是由与st25dv共用天线的ic芯片实现的，这两个洞的确要加芯片才行，不直接模拟的原因是一个是iso15693，一个是iso14443，标准不同，下次发issue记得先看之前的

jimianjun, 2020-07-04:
> 目前up的版本是添加两张卡，是由与st25dv共用天线的ic芯片实现的，这两个洞的确要加芯片才行，不直接模拟的原因是一个是iso15693，一个是iso14443，标准不同，下次发issue记得先看之前的

我理解是这样的，st25dv的作用是跟手机nfc传输图像显示在显示屏上面是吗？那是他、25dv的储存大小，04k，16k，64k，这大小有什么用呢？那两个洞放的就是普通门禁卡里面的那芯片吗？可以uid，cuid，这种芯片？你说的iso 15693，14443，两种协议是什么意思，st25支持两种协议，还是洞支持一个15693一个14443，还是芯片协议随便放只要不是ID，我是小白啊，麻烦你了

iwtywai, 2020-07-04:
st25dv在整个板子上应该就是传输数据的，容量大小我没太明白，up说选哪种都行。那两个洞放的就是普通门禁卡里面的那芯片，可以uid，cuid。iso 15693是25dv的协议，14443是uid芯片的，两种协议不一样，所以不能直接只用25dv。

jimianjun, 2020-07-04:
> st25dv在整个板子上应该就是传输数据的，容量大小我没太明白，up说选哪种都行。那两个洞放的就是普通门禁卡里面的那芯片，可以uid，cuid。iso 15693是25dv的协议，14443是uid芯片的，两种协议不一样，所以不能直接只用25dv。

懂了，就是这card是用的st25dv芯片，只能传输 15693的协议，也只能添加两个15693芯片的uid卡，cuid卡，普通s50卡，可不可以直接用这个芯片生成 模拟15693协议的门禁卡，就跟手机一样，可以随便生成几张门禁卡，手机的限制就是卡号不可更改，直接用芯片生成模拟卡

iwtywai, 2020-07-04:
> > st25dv在整个板子上应该就是传输数据的，容量大小我没太明白，up说选哪种都行。那两个洞放的就是普通门禁卡里面的那芯片，可以uid，cuid。iso 15693是25dv的协议，14443是uid芯片的，两种协议不一样，所以不能直接只用25dv。
> 
> 懂了，就是这card是用的st25dv芯片，只能传输 15693的协议，也只能添加两个15693芯片的uid卡，cuid卡，普通s50卡，可不可以直接用这个芯片生成 模拟15693协议的门禁卡，就跟手机一样，可以随便生成几张门禁卡，手机的限制就是卡号不可更改，直接用芯片生成模拟卡

还不是，那uid芯片是14443，和正常卡一样把卡信息刷进去，直接拿读写器。至于要拿25dv模拟15693，我不确定行不行，还没看过文档，up只是用了个25dv的功能。这和手机的随便生成还是不一样的，这个设备目前只能在两张卡之间切换，多张卡+直接模拟还要等up出第二版。

jimianjun, 2020-07-04:
懂了懂了

jimianjun, 2020-07-04:
> 还不是，那uid芯片是14443，和正常卡一样把卡信息刷进去，直接拿读写器。至于要拿25dv模拟15693，我不确定行不行，还没看过文档，up只是用了个25dv的功能。这和手机的随便生成还是不一样的，这个设备目前只能在两张卡之间切换，多张卡+直接模拟还要等up出第二版。

st25d芯片和ic卡芯片其实是独立的，只是共用了一个天线而已，所有14443和15693其实都可以，只要你的读卡器支持对吧，拨轮开关的作用就是两张卡之间切换而已

iwtywai, 2020-07-04:
对，但我不知道15693的芯片能不能用，up用的14443，我个人觉得可以

guanzi008, 2020-07-23:
有没有成品做出来的，我现在一头雾水。
有没有愿意出售成品的我的wx是odatacc

feichaijiemi, 2020-07-23:
进群：144538834

XiaobaoAlex, 2020-08-18:
> 还不是，那uid芯片是14443，和正常卡一样把卡信息刷进去，直接拿读写器。至于要拿25dv模拟15693，我不确定行不行，还没看过文档，up只是用了个25dv的功能。这和手机的随便生成还是不一样的，这个设备目前只能在两张卡之间切换，多张卡+直接模拟还要等up出第二版。

我理解是这样，25dv只是用来接收手机传来的的信息的，接收的信息在墨水屏上显示，25dv通过拨盘和两个uid芯片共用天线,实际上如果只是需要门禁功能，只要两个uid芯片和天线就行了，是这样嘛？

iwtywai, 2020-08-18:
是的，我也试了一下，只装上nfc芯片然后用跳线连接贴片拨轮的焊盘是可以使用的

syf2005, 2022-04-12:
交通一卡通大概率是CPU模拟卡，基本不可能破解写入uid卡片

---- 在 星期四, 2022-04-07 17:33:24 张瑞鑫 ***@***.***> 写到 ----

是的，我也试了一下，只装上nfc芯片然后用跳线连接贴片拨轮的焊盘是可以使用的
 uid芯片怎么写入其他卡片的信息呢？只用两个uid芯片模拟两个卡片，比如说模拟我的交通一卡通。怎么用两个小uid芯片模拟自己的交通一卡通。