peng-zhihui/L-ink_Card #64: 生成Hex报错

BH6AOV, 2020-07-18:
Keil5生成Hex时报错了，报错如下
```
Rebuild started: Project: L-ink_Card
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\armkeil\ARM\ARMCC\Bin'
Rebuild target 'L-ink_Card'
assembling startup_stm32l051xx.s...
startup_stm32l051xx.s: Error: A9555E: Failed to check out a license.LICENSE ERROR (R207(2): REGISTRY READ ERROR)
startup_stm32l051xx.s: 
startup_stm32l051xx.s: Check that your license details are correct in the License Management dialog of MDK. Additional information is available at:
startup_stm32l051xx.s:  http://www.keil.com/support/man/docs/license/license_management.htm
startup_stm32l051xx.s:  If you need further help, provide this complete error report to your supplier or license.support@arm.com.
startup_stm32l051xx.s:  - ARMLMD_LICENSE_FILE: unset
startup_stm32l051xx.s:  - LM_LICENSE_FILE: unset
startup_stm32l051xx.s:  - ARM_TOOL_VARIANT: unset
startup_stm32l051xx.s:  - ARM_PRODUCT_PATH: unset
startup_stm32l051xx.s:  - Product location: C:\Keil_v5\armkeil\ARM\sw\mappings
startup_stm32l051xx.s:  - Toolchain location: C:\Keil_v5\armkeil\ARM\ARMCC\Bin
startup_stm32l051xx.s:  - Selected tool variant: mdk_std
startup_stm32l051xx.s:  - Checkout feature: LIC0=KA...-.....-.....
startup_stm32l051xx.s:  - Feature version: 5.0201709
startup_stm32l051xx.s:  - Keil error code: 1
startup_stm32l051xx.s: Product: MDK Plus 5.31
startup_stm32l051xx.s: Component: ARM Compiler 5.06 update 6 (build 750)
startup_stm32l051xx.s: Tool: ArmAsm [4d35ec]
compiling main.c...
Error: C9555E: Failed to check out a license.LICENSE ERROR (R207(2): REGISTRY READ ERROR)
Check that your license details are correct in the License Management dialog of MDK. Additional information is available at:
 http://www.keil.com/support/man/docs/license/license_management.htm
 If you need further help, provide this complete error report to your supplier or license.support@arm.com.
 - ARMLMD_LICENSE_FILE: unset
 - LM_LICENSE_FILE: unset
 - ARM_TOOL_VARIANT: unset
 - ARM_PRODUCT_PATH: unset
 - Product location: C:\Keil_v5\armkeil\ARM\sw\mappings
 - Toolchain location: C:\Keil_v5\armkeil\ARM\ARMCC\Bin
 - Selected tool variant: mdk_std
 - Checkout feature: LIC0=KA...-.....-.....
 - Feature version: 5.0201709
 - Keil error code: 1
Product: MDK Plus 5.31
Component: ARM Compiler 5.06 update 6 (build 750)
Tool: ArmCC [4d3637]
compiling gpio.c...
Error: C9555E: Failed to check out a license.LICENSE ERROR (R207(2): REGISTRY READ ERROR)
Check that your license details are correct in the License Management dialog of MDK. Additional information is available at:
 http://www.keil.com/support/man/docs/license/license_management.htm
 If you need further help, provide this complete error report to your supplier or license.support@arm.com.
 - ARMLMD_LICENSE_FILE: unset
 - LM_LICENSE_FILE: unset
 - ARM_TOOL_VARIANT: unset
 - ARM_PRODUCT_PATH: unset
 - Product location: C:\Keil_v5\armkeil\ARM\sw\mappings
 - Toolchain location: C:\Keil_v5\armkeil\ARM\ARMCC\Bin
 - Selected tool variant: mdk_std
 - Checkout feature: LIC0=KA...-.....-.....
 - Feature version: 5.0201709
 - Keil error code: 1
Product: MDK Plus 5.31
Component: ARM Compiler 5.06 update 6 (build 750)
Tool: ArmCC [4d3637]
compiling usart.c...
Error: C9555E: Failed to check out a license.LICENSE ERROR (R207(2): REGISTRY READ ERROR)
Check that your license details are correct in the License Management dialog of MDK. Additional information is available at:
 http://www.keil.com/support/man/docs/license/license_management.htm
 If you need further help, provide this complete error report to your supplier or license.support@arm.com.
 - ARMLMD_LICENSE_FILE: unset
 - LM_LICENSE_FILE: unset
 - ARM_TOOL_VARIANT: unset
 - ARM_PRODUCT_PATH: unset
 - Product location: C:\Keil_v5\armkeil\ARM\sw\mappings
[... 491 lines cut ...]
Product: MDK Plus 5.31
Component: ARM Compiler 5.06 update 6 (build 750)
Tool: ArmCC [4d3637]
compiling system_stm32l0xx.c...
Error: C9555E: Failed to check out a license.LICENSE ERROR (R207(2): REGISTRY READ ERROR)
Check that your license details are correct in the License Management dialog of MDK. Additional information is available at:
 http://www.keil.com/support/man/docs/license/license_management.htm
 If you need further help, provide this complete error report to your supplier or license.support@arm.com.
 - ARMLMD_LICENSE_FILE: unset
 - LM_LICENSE_FILE: unset
 - ARM_TOOL_VARIANT: unset
 - ARM_PRODUCT_PATH: unset
 - Product location: C:\Keil_v5\armkeil\ARM\sw\mappings
 - Toolchain location: C:\Keil_v5\armkeil\ARM\ARMCC\Bin
 - Selected tool variant: mdk_std
 - Checkout feature: LIC0=KA...-.....-.....
 - Feature version: 5.0201709
 - Keil error code: 1
Product: MDK Plus 5.31
Component: ARM Compiler 5.06 update 6 (build 750)
Tool: ArmCC [4d3637]
"L-ink_Card\L-ink_Card.axf" - 32 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:06
```

iwtywai, 2020-07-18:
它说是读不了注册表，看看能不能管理员模式运行，或者重新安装、注册一下软件

BH6AOV, 2020-07-18:
> 它说是读不了注册表，看看能不能管理员模式运行，或者重新安装、注册一下软件

了解了，我试一下管理员模式

BH6AOV, 2020-07-18:
> 它说是读不了注册表，看看能不能管理员模式运行，或者重新安装、注册一下软件

并没有什么区别。。。软件前几天才装的最新版，也是才注册的。。。不应该啊

peng-zhihui, 2020-07-18:
keil的license到期了，需要重新激活，激活需要在管理员模式下进行。

BH6AOV, 2020-07-18:
> keil的license到期了，需要重新激活，激活需要在管理员模式下进行。

看了下我激活的那个证书写的到2024年11月才过期啊管理员模式装的，不过是注册机生成的

iwtywai, 2020-07-18:
我记得203几年的是没有问题的，找找看那个试试？

BH6AOV, 2020-07-18:
> 我记得203几年的是没有问题的，找找看那个试试？

我没有(﹁"﹁)方便给个链不

iwtywai, 2020-07-18:
没存...😭

BH6AOV, 2020-07-18:
> 没存...😭

😭我再找找

yelvlab, 2020-08-27:
> > 没存...😭
> 
> 😭我再找找

https://blog.csdn.net/Mculover666/article/details/89469764
我重新装的531，好久没用过keil了，我用jlink烧了，安卓应用也提示正在传输图片，然后后续就没有任何反应了，头大

closed 2025-01-20 by BH6AOV