peng-zhihui/L-ink_Card #79: 【求助】墨水屏没有一点点反应，不知该如何下手

yelvlab, 2020-08-27:
拖了几个月终于焊好了，然后装的531的keil，PJ到了2032年，成功用jlink烧录

```
Build started: Project: L-ink_Card
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'd:\Keil_v5\ARM\ARMCC\Bin'
Build target 'L-ink_Card'
"L-ink_Card\L-ink_Card.axf" - 0 Error(s), 0 Warning(s).
Build Time Elapsed:  00:00:00
Load "L-ink_Card\\L-ink_Card.axf" 
Set JLink Project File to "D:\user\hulk\Desktop\Keil\L-ink_Card\MDK-ARM\JLinkSettings.ini"
* JLink Info: Device "STM32L051K8" selected.
 
JLink info:
------------
DLL: V6.70e, compiled Apr 17 2020 17:55:05
Firmware: J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
Hardware: V7.00
S/N : 20140522 
Feature(s) : RDI,FlashDL,FlashBP,JFlash,GDBFull 
 
* JLink Info: Found SW-DP with ID 0x0BC11477
* JLink Info: Found SW-DP with ID 0x0BC11477
* JLink Info: Unknown DP version. Assuming DPv0
* JLink Info: Scanning AP map to find all available APs
* JLink Info: AP[1]: Stopped AP scan as end of AP map has been reached
* JLink Info: AP[0]: AHB-AP (IDR: 0x04770031)
* JLink Info: Iterating through AP map to find AHB-AP to use
* JLink Info: AP[0]: Core found
* JLink Info: AP[0]: AHB-AP ROM base: 0xF0000000
* JLink Info: CPUID register: 0x410CC601. Implementer code: 0x41 (ARM)
* JLink Info: Found Cortex-M0 r0p1, Little endian.
* JLink Info: FPUnit: 4 code (BP) slots and 0 literal slots
* JLink Info: CoreSight components:
* JLink Info: ROMTbl[0] @ F0000000
* JLink Info: ROMTbl[0][0]: E00FF000, CID: B105100D, PID: 000BB4C0 ROM Table
* JLink Info: ROMTbl[1] @ E00FF000
* JLink Info: ROMTbl[1][0]: E000E000, CID: B105E00D, PID: 000BB008 SCS
* JLink Info: ROMTbl[1][1]: E0001000, CID: B105E00D, PID: 000BB00A DWT
* JLink Info: ROMTbl[1][2]: E0002000, CID: B105E00D, PID: 000BB00B FPB
ROMTableAddr = 0xF0000000
* JLink Info: Reset: Halt core after reset via DEMCR.VC_CORERESET.
* JLink Info: Reset: Reset device via AIRCR.SYSRESETREQ.
 
Target info:
------------
Device: STM32L051K8Tx
VTarget = 3.300V
State of Pins: 
TCK: 0, TDI: 0, TDO: 1, TMS: 0, TRES: 1, TRST: 1
Hardware-Breakpoints: 4
Software-Breakpoints: 8192
Watchpoints:          2
JTAG speed: 4000 kHz
 
Erase Done.
Programming Done.
Verify OK.
Flash Load finished at 17:40:05
```

然后我的墨水屏没有任何反应，LED也没有，代码一知半解的，看不太懂，不知道该如何下手，两个IC芯片已经没问题了，可以读写，然后ST25那个NFC芯片手机有反应，打开L-ink那个安卓应用，二值化、模糊处理和标签那些东西手机只会提示一个”正在传输图片，请不要拿走卡片“，墨水屏和LED还是没有反应。

如上所述，我该从哪里入手，或者有没有大佬给一份直接墨水屏亮屏的代码，LED我尝试凭借记忆写了一下，也没反应，我也不知道是我没写对还是芯片有问题(正常来说，keil的debugger里面的SWD可以识别，芯片应该是正常的)。

huajijam, 2020-08-27:
同无反应，已确定屏幕为V2 两块板子两块屏幕都测试过了，元件也测试过了，但我是能传输成功但是传输成功之后无显示

yelvlab, 2020-08-28:
> 同无反应，已确定屏幕为V2 两块板子两块屏幕都测试过了，元件也测试过了，但我是能传输成功但是传输成功之后无显示

我也是V2屏幕

yelvlab, 2020-08-28:
> 同无反应，已确定屏幕为V2 两块板子两块屏幕都测试过了，元件也测试过了，但我是能传输成功但是传输成功之后无显示

老哥，你的问题好了么，我的已经好了，用了另外一个大佬改过的代码，可能有些差异，不适配

closed 2020-08-28 by yelvlab

huajijam, 2020-08-29:
> > 同无反应，已确定屏幕为V2 两块板子两块屏幕都测试过了，元件也测试过了，但我是能传输成功但是传输成功之后无显示
> 
> 老哥，你的问题好了么，我的已经好了，用了另外一个大佬改过的代码，可能有些差异，不适配

哪个？给个链？

yelvlab, 2020-08-31:
> > > 同无反应，已确定屏幕为V2 两块板子两块屏幕都测试过了，元件也测试过了，但我是能传输成功但是传输成功之后无显示
> > 
> > 
> > 老哥，你的问题好了么，我的已经好了，用了另外一个大佬改过的代码，可能有些差异，不适配
> 
> 哪个？给个链？

老哥，加一下群，我没找到github地址，群里一位大佬的  144538834