peng-zhihui/L-ink_Card #94: 如果我不需要模拟门禁卡的功能，只需要用一块墨水瓶，然后加上NFC的功能呢？

12121-12121, 2021-01-06:
这样可能有点偏离初衷了，，，
但我是这样想的，借题发挥吧。如果我不需要模拟门禁卡的功能，只需要一块墨水瓶加上NFC功能，实现用手机一碰就能把手机里的照片传到墨水瓶上，功耗问题应该不用着重考虑，请问这样该怎么实现呢，用树莓派或者arduino行吗？pcb我实在不会画，还有我从网上百度了，没人做这个，都是做相册什么的。
我是初中生，刚接触这方面，水平还不够，请多多指教！

12121-12121, 2021-01-06:
> 这样可能有点偏离初衷了，，，
> 但我是这样想的，借题发挥吧。如果我不需要模拟门禁卡的功能，只需要一块墨水屏加上NFC功能，实现用手机一碰就能把手机里的照片传到墨水瓶上，功耗问题应该不用着重考虑，请问这样该怎么实现呢，用树莓派或者arduino行吗？pcb我实在不会画，还有我从网上百度了，没人做这个，都是做相册什么的。
> 我是初中生，刚接触这方面，水平还不够，请多多指教！

fml927, 2021-01-26:
树莓派加个显示器和NFC设备 应该就，只是得自己捣鼓下

Yinke7, 2021-03-09:
微雪官网上有你这种，一个墨水屏加NFC的，指路：https://www.waveshare.net/shop/1.54inch-NFC-Powered-e-Paper-BW.htm

csr865, 2021-05-03:
> 这样可能有点偏离初衷了,,,
> 但我是这样想的，借题发挥吧。如果我不需要模拟门禁卡的功能，只需要一块墨水瓶加上NFC功能，实现用手机一碰就能把手机里的照片传到墨水瓶上，导致问题应该不用着重考虑，请问这样该怎么实现呢，用树莓派或者arduino行吗？pcb我实在不会画，还有我从网上百度了，没人做这个，都是做相册什么的。
> 我是初中生，刚接触这方面，水平还不够，请多多指教！

确实而且对结构差别并不大，就加了两个UID芯片而已