peng-zhihui/L-ink_Card #101: 基于L-ink_card的改进版<目前的问题>求帮助

lyjuestc, 2021-02-07:
利用st95来做软件层面的模拟卡，同时实现对nfc卡的读取。st95正好支持多种协议的卡片读写功能和卡片模拟功能。

目前的问题：参考st95官方评估板做了一套板子，能够读取nfc卡的uid，但是无法读取卡的扇区数据，错误返回为传输错误，但是能够读取uid，读写芯片寄存器都可以成功。

有没有大佬做过类似的NFC项目，指导一下。。。。

这是st95评估板代码中的数据读取代码。运行到这里每次都是error。
[image]

[image]

求帮助啊、、。。。。。。

Yinke7, 2021-03-09:
Sorry, your problem is too simple. I hope it will be better for you to solve it  by yourself

HouqiLi, 2022-04-12:
> 利用st95来做软件层面的模拟卡，同时实现对nfc卡的读取。st95正好支持多种协议的卡片读写功能和卡片模拟功能。
> 
> 目前的问题：参考st95官方评估板做了一套板子，能够读取nfc卡的uid，但是无法读取卡的扇区数据，错误返回为传输错误，但是能够读取uid，读写芯片寄存器都可以成功。
> 
> 有没有大佬做过类似的NFC项目，指导一下。。。。
> 
> 这是st95评估板代码中的数据读取代码。运行到这里每次都是error。 [image]
> 
> [image]
> 
> 求帮助啊、、。。。。。。

你好，请问有ST95的demo吗？你的问题解决没有？

wdon9, 2022-07-21:
同问

wdon9, 2022-07-21:
@lyjuestc HELLO，问题解决了吗？