peng-zhihui/L-ink_Card #103: Cilon下载不成功，DapLink和ST-Link V2都试过了，显示

Tony-Tang0723, 2021-02-18:
F:\OpenOCD\openocd-20201228\OpenOCD-20201228-0.10.0\bin\openocd.exe -c "tcl_port disabled" -s F:\OpenOCD\openocd-20201228\OpenOCD-20201228-0.10.0\share\openocd\scripts -c "gdb_port 3333" -c "telnet_port 4444" -f board\st_nucleo_f103rb.cfg -c "program \"C:/tony/L-ink_Card-master/L-ink_Card-master/Firmware/Clion/L-ink_Card/cmake-build-debug/L-ink_Card.elf\"" -c reset -c shutdown
[0m[0mOpen On-Chip Debugger 0.10.0 (2020-12-28) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlin
k.cfg
Info : The selected transport took over low-level target control. The results mi
ght differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

求一个不会出错的.cfg文件，谢谢

sparrowhe, 2021-07-10:
try：https://community.platformio.org/t/openocd-init-failed/12689

Asensio-ai, 2021-07-22:
您好，请问问题解决了吗？我也遇到了相同的问题

ab1324ab, 2022-06-11:
我的能下载 但不能调试