peng-zhihui/L-ink_Card #115: 关于TP6122

xiaokamikami, 2021-11-24:
TI的TP6122没货了可以换成SX1308 B628 升压芯片吗

netube99, 2022-01-24:
很明显不行，1308在小电流输出时效率很低的

netube99, 2022-01-24:
> TI的TP6122没货了可以换成SX1308 B628 升压芯片吗

很明显不行，1308小电流输出时效率太低，不符合低功耗要求

jiuchegnbanxia, 2022-03-21:
> > TI的TP6122没货了可以换成SX1308 B628 升压芯片吗
> 
> 直接把TPS61221去掉吧，还有升压电感，直接把输入和输出用导线连接起来，通过电池供电就行了。可以把电池电压用到二伏多一点。你看原理图会发现升压电路是错的，没有了输入和输出电容，所以无法完成升压的，另外电源部分的PCB布局也有些问题。可能是当初设计时没仔细看文档吧，上面明明说的很清楚的。

原理图没错啊，这个芯片就是固定升压输出3.3V，干电池作为输入端，而且电路中没有放大器、振荡器，输入端和输出端的滤波电容是可以不加的

Libaididi, 2022-04-12:
tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。

GYSS1204, 2022-04-12:
> > > TI的TP6122没货了可以换成SX1308 B628 升压芯片吗
> > 
> > 直接把TPS61221去掉吧，还有升压电感，直接把输入和输出用导线连接起来，通过电池供电就行了。可以把电池电压用到二伏多一点。你看原理图会发现升压电路是错的，没有了输入和输出电容，所以无法完成升压的，另外电源部分的PCB布局也有些问题。可能是当初设计时没仔细看文档吧，上面明明说的很清楚的。
> 
> 原理图没错啊，这个芯片就是固定升压输出3.3V，干电池作为输入端，而且电路中没有放大器、振荡器，输入端和输出端的滤波电容是可以不加的

可是这并不是基于电荷泵的升压电路，而是开关直流升压电路，你可以看下boost升压电路原理，电容和电感是升压的关键元件。

Libaididi, 2022-04-12:
[image]这是典型应用电路。

GYSS1204, 2022-04-12:
> [image]这是典型应用电路。
> 

是的，作者的原理图里缺少了电容，所以导致不能正常升压。你可以去除这个芯片，并且把pcb上芯片vin和vout的焊盘短接，l-ink是可以正常工作的，这是最方便的一种解决办法。

jiuchegnbanxia, 2022-04-12:
> 可是这并不是基于电荷泵的升压电路，而是开关直流升压电路，你可以看下boost升压电路原理，电容和电感是升压的关键元件。
开关直流升压电路电感给电容充电实现升压，确实都是关键器件。输入电容为了抗干扰、改善瞬态响应，应该也能去，输出端的电容按理说是升压电容，但是我按他的电路试了是可以输出3.3V的（空载），我回头把输出电容接上对比下带载的时候是啥样

jiuchegnbanxia, 2022-04-12:
> tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。

可替代的太多了，我用的RT9193-33GB

GYSS1204, 2022-04-12:
> 开关直流升压电路电感给电容充电实现升压，确实都是关键器件。输入电容为了抗干扰、改善瞬态响应，应该也能去，输出端的电容按理说是升压电容，但是我按他的电路试了是可以输出3.3V的（空载），我回头把输出电容接上对比下带载的时候是啥样

额，那你可以测量下新的纽扣电池电压，是在3.3v左右。或者你有示波器的话测量电感，看下是否有开关电压波形。

GYSS1204, 2022-04-12:
> > tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。
> 
> 可替代的太多了，我用的RT9193-33GB

RT9193-33GB是线性降压稳压芯片，没法升压，所以不适合用在这里。

GYSS1204, 2022-04-12:
> tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。

具体你可以看下电荷泵升压的芯片xa2320b和hx4002b，不过需要你重新设计电路修改pcb板，不能直接在作者的pcb上使用。

jiuchegnbanxia, 2022-04-12:
> > tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。
> 
> 具体你可以看下电荷泵升压的芯片xa2320b和hx4002b，不过需要你重新设计电路修改pcb板，不能直接在作者的pcb上使用。

> > > tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。
> > 
> > 
> > 可替代的太多了，我用的RT9193-33GB
> 
> RT9193-33GB是线性降压稳压芯片，没法升压，所以不适合用在这里。

嗯嗯，这个知道，我没说清楚，我是换了3.7V的锂电

jiuchegnbanxia, 2022-04-12:
> > tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。
> 
> 具体你可以看下电荷泵升压的芯片xa2320b和hx4002b，不过需要你重新设计电路修改pcb板，不能直接在作者的pcb上使用。

好的，我回头瞅瞅看好不好买

jiuchegnbanxia, 2022-04-13:
> tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。

原方案的话，我又找了下，可以用航天民芯的MT3540

Libaididi, 2022-04-13:
> > tps61221有没有替代方案，这个芯片淘宝卖60根本买不起。
> 
> 原方案的话，我又找了下，可以用航天民芯的MT3540
你好，把这个芯片直接焊到tps6122的位置就可以了吗？

jiuchegnbanxia, 2022-04-13:
> 
不可以，需要按手册重新设计电路