peng-zhihui/L-ink_Card #123: 一个纯小白想请教下各位大佬

G1kl, 2022-09-06:
刚用github，如果这个内容不属于issue的话我把他关掉（如果可以的话）
其实B站的视频我看下来感觉稚晖君在说他的设计思路，我想要搞懂几个主要原件的作用
元器件的焊接好像可以在打板子的时候一起焊？自己焊的话对技术要求高吗？我自己以前用过电烙铁也焊过一点点东西，但是技术自认为挺差的
我看到板子中间挖了两个孔，相当于这个设计并没有把卡里面的数据复制下来，而是需要自己把卡里面的芯片取出之后焊上去吗？
关于烧录，烧录指的就是把那些源代码写入芯片里面的步骤吗？自己在电脑上烧录的话除了需要在淘宝购买相关烧录的软件（我看有些issue下面是这么说的）还需要什么东西吗？感觉看github上的过程也有点蒙

Rulsion, 2022-09-06:
已收到，放心

Mahaochen11, 2022-10-27:
我也想问，卡多的话咋办

bicohuang, 2023-02-26:
> 刚用github，如果这个内容不属于issue的话我把他关掉（如果可以的话） 其实B站的视频我看下来感觉稚晖君在说他的设计思路，我想要搞懂几个主要原件的作用 元器件的焊接好像可以在打板子的时候一起焊？自己焊的话对技术要求高吗？我自己以前用过电烙铁也焊过一点点东西，但是技术自认为挺差的 我看到板子中间挖了两个孔，相当于这个设计并没有把卡里面的数据复制下来，而是需要自己把卡里面的芯片取出之后焊上去吗？ 关于烧录，烧录指的就是把那些源代码写入芯片里面的步骤吗？自己在电脑上烧录的话除了需要在淘宝购买相关烧录的软件（我看有些issue下面是这么说的）还需要什么东西吗？感觉看github上的过程也有点蒙

中间两个孔是用来放芯片。稚晖君他自己放是网上买的空白芯片，理论上是可以剪卡取芯片放上去，所谓烧录，是指把程序写入指定的芯片的，在这里是指把编译好的程序写入的stm32L0芯片的过程，一般情况下需要一个烧录专用的设备就行了，很多ide都集成烧录指令

Rulsion, 2023-02-26:
已收到，放心